chenxiaolong/MSD #11: Attempting to enable mass storage functions throws an error.

hax4dazy, 2024-10-06:
After checking the readme and looking if my device has all the functions that the app needs, it seemingly still does not want to enable the mass storage function on my Samsung Galaxy S8 running LineageOS 21. The app throws a `Failed to get list of active USB gadget functions` with the error details being `IOException (Connection Refused)`

Here is a screenshot of all the features that my device should have as specified from the readme as I am not really sure what to look for.
[image]

chenxiaolong, 2024-10-06:
Thanks for posting the command output! Can you also pull `/data/local/tmp/msd.post-fs-data.log` and `/data/local/tmp/msd.service.log` and upload those 2 logs?

At first glance, it seems that your device doesn't use `android.hardware.usb.gadget-service`, most likely because it originally shipped with an Android version older than 12. It might still be possible to make MSD work, but I'll need to poke around and see what LineageOS 21 does for managing USB gadget functionality on older devices.

hax4dazy, 2024-10-07:
Here is the log for [msd.post-fs-data.log](https://pastebin.com/bgMMeYcJ) and here's the log for [msd.service.log](https://pastebin.com/s1Bs6EqL) 

The phone shipped with android 8 (I think) from the factory. Also keep in mind that I'm using [this](https://xdaforums.com/t/lineageos-21-0-android-14-signature-spoofing-ota-updates-for-s8-s8-note8.4655365/) custom rom as the Galaxy S8 isn't officially supported yet by lineage themselves

sethfoxen, 2024-10-08:
I'm also having this issue on my Google Pixel 6 Pro, running Lineage OS 21. For what it's worth, I've used the "isodrive" (https://github.com/nitanmarcel/isodrive-magisk) module to great success. I'll attach the same files as @hax4dazy for you to check out. 

[image]
[msd.post-fs-data.log](https://github.com/user-attachments/files/17285794/msd.post-fs-data.log)
[msd.service.log](https://github.com/user-attachments/files/17285796/msd.service.log)

chenxiaolong, 2024-10-08:
@sethfoxen Thanks for the logs! While the symptoms are similar, the root cause is different for your device, so I've moved it to #12.

chenxiaolong, 2024-10-08:
@hax4dazy Thanks for the logs and the link to the ROM! That'll be helpful for figuring this out.

chenxiaolong, 2024-10-08:
@hax4dazy Can you post the output of these 3 commands as well?

```bash
ps -efZ | grep -i usb
getprop sys.usb.config
getprop sys.usb.configfs
```

hax4dazy, 2024-10-08:
Alright, here's the output of those three commands:

[image]

As for the command in the edit, I spammed it in a CMD window while booting the phone up and having it plugged into my computer. [This](https://pastebin.com/8BTeN181) is the output of that command as soon as ADB stopped saying that there weren't any devices/emulators found. Hope this helps!

chenxiaolong, 2024-10-08:
Thanks! The `sys.usb.configfs=2` is weird and probably a bug in the ROM, but no big deal. That confirmed what I needed to know: LineageOS on your device manages the USB mode using init scripts instead of Android 12+'s USB gadget daemon.

---

So the next thing we need to figure out is why this is happening (from your `msd.service.log`):
```
runcon: exec /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a: Permission denied
```

Unfortunately, https://pastebin.com/8BTeN181 did not contain what I was expecting. Can you run:
```bash
ls -lZ /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a
```
and check if the output matches this?
```
-rwxr-xr-x 1 root root u:object_r:system_file:s0  1592496 2024-08-05 23:35 /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a
```
(specifically, the `rwxr-xr-x` permissions and `u:object_r:system_file:s0` SELinux label)

---

If that all looks good, please run:

```
dmesg -w | grep avc
```

and then in a separate terminal window, run:

```
/system/bin/runcon u:r:msd_daemon:s0 /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a daemon --log-level debug
```

After the second command fails, I'd expect to see some new `denied` messages from `dmesg`. (Though sometimes, it might take a minute to show up.)

hax4dazy, 2024-10-09:
> and check if the output matches this?

```
-rwxr-xr-x 1 root root u:object_r:system_file:s0  1592496 2024-10-06 12:17 /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a
```
Seems like it matches up correctly. Date is different but I kinda expect that to be since different rom and stuff

---

I also ran the command you said to run in another terminal window while monitoring it with the first one and it showed me nothing new. I waited like about 15ish minutes and even ran the command a second time but it produced no new dmesg messages
Here's the output anyways, just in case it could be useful

```
dreamlte:/ # dmesg -w | grep avc
[    8.494114] SELinux: avc:  op=load_policy lsm=selinux seqno=2 res=1
[    8.700598] SELinux: avc:  op=load_policy lsm=selinux seqno=2 res=1
[    9.131181] SELinux: avc:  denied  { find } for pid=4916 uid=1021 name=android.system.suspend.ISystemSuspend/default scontext=u:r:lhd:s0 tcontext=u:object_r:hal_system_suspend_service:s0 tclass=service_manager permissive=1
[   16.185858] SELinux: avc:  denied  { find } for pid=4917 uid=1021 name=android.system.suspend.ISystemSuspend/default scontext=u:r:gpsd:s0 tcontext=u:object_r:hal_system_suspend_service:s0 tclass=service_manager permissive=1
```

chenxiaolong, 2024-10-10:
Hmm, that's really odd. In the second terminal, did it fail with this error:
```
runcon: exec /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a: Permission denied
```
or something different? (or did it stay running and produce no output at all?)

---

Please give this test build a try: [MSD-1.3.r5.g6c9862f-release.zip](https://github.com/user-attachments/files/17321870/MSD-1.3.r5.g6c9862f-release.zip)

If it doesn't work, please upload the `/data/local/tmp/msd.post-fs-data.log` and `/data/local/tmp/msd.service.log` logs like before and also `/sys/fs/selinux/policy`. For the policy file, you might need to put it in a zip for Github to allow it to be uploaded.

hax4dazy, 2024-10-10:
Yes it failed with that exact error I think. 

---

I tried the new build and it still fails with the same error as before sadly. Here are the logs you asked for + the policy file. The policy file is hosted somewhere else then GitHub since it didn't want to be hosted on here (also it was a pain to pull off of my phone lol)

[msd.service.log](https://github.com/user-attachments/files/17330808/msd.service.log)
[msd.post-fs-data.log](https://github.com/user-attachments/files/17330809/msd.post-fs-data.log)
[policy](https://files.catbox.moe/swk3xj)

chenxiaolong, 2024-10-11:
Thanks for bearing with me and providing everything I've been asking for!

Looking at the policy, everything is correct, so I don't think this is an SELinux issue.

I've been digging into the LineageOS build for the past couple hours and I think I found the problem. I extracted the kernel configuration from `boot.img` inside `lineage-21.0-20240820-UNOFFICIAL-dreamlte.zip`. It was compiled with these options set:

```
#
# Samsung Rooting Restriction Feature
#
CONFIG_SEC_RESTRICT_ROOTING=y
CONFIG_SEC_RESTRICT_SETUID=y
CONFIG_SEC_RESTRICT_FORK=y
CONFIG_SEC_RESTRICT_ROOTING_LOG=y
```

Normally, custom ROMs disable these. They're part of the stock ROM's restrictions on root functionality.

I found the source code for the custom ROM's kernel and it looks like they're not explicitly setting these config options, so they fall back to the default value (yes): https://github.com/8890q/android_kernel_samsung_universal8895/blob/d93f7bb6e2aa46ba4889796f0d8d79e95279150d/arch/arm64/configs/exynos8895-dreamlte_defconfig#L5223-L5226

These options prevent files in `/data` from running as root, which I think is why you keep getting `permission denied` when MSD runs. I don't know any good way to work around this. I think the custom ROM will need to be updated to disable these restrictions.

ivanmeler, 2024-10-11:
> Thanks for bearing with me and providing everything I've been asking for!
>
> Looking at the policy, everything is correct, so I don't think this is an SELinux issue.
>
> I've been digging into the [...]

false positive on that one, defconfig used is one you see on github this config.gz is coming from
https://github.com/8890q/android_kernel_samsung_universal8895/commit/48b56858e40addb8412e41f8d10c8a6b5e2904e4