chenxiaolong/MSD #12: Cannot connect to daemon on Pixel 6 Pro

chenxiaolong, 2024-10-08:
> I'm also having this issue on my Google Pixel 6 Pro, running Lineage OS 21. For what it's worth, I've used the "isodrive" (https://github.com/nitanmarcel/isodrive-magisk) module to great success. I'll attach the same files as @hax4dazy for you to check out. 
>
> [image]
> [msd.post-fs-data.log](https://github.com/user-attachments/files/17285794/msd.post-fs-data.log)
> [msd.service.log](https://github.com/user-attachments/files/17285796/msd.service.log)

_Originally posted by @sethfoxen in https://github.com/chenxiaolong/MSD/issues/11#issuecomment-2398287449_

chenxiaolong, 2024-10-08:
@sethfoxen Can you reproduce the problem and then immediately run this command to grab the logs for the daemon?

```bash
adb logcat -s msd-tool -d msd-tool.logcat
```

Also, do you get any errors when running these commands as root?

```bash
/data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a client get-functions

/data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a client set-mass-storage /sdcard/YOUR_IMAGE_FILE.img
```

sethfoxen, 2024-10-08:
Simply tapping the "Enable mass storage function" causes the error. Here are the results. 

> ```shell
> adb logcat -s msd-tool -d msd-tool.logcat
> ```

` adb logcat -s msd-tool -d msd-tool.logcat
* daemon not running; starting now at tcp:5037
* daemon started successfully
--------- beginning of main
10-08 10:26:30.821  1672 14254 I msd-tool: peer{pid=11504 uid=10250 gid=10250}: msd_tool::daemon: Received connection
10-08 10:26:30.953  1672 14254 W msd-tool: peer{pid=11504 uid=10250 gid=10250}: msd_tool::daemon: Failed to find gadget HAL process
10-08 10:26:30.961  1672 14281 I msd-tool: peer{pid=11504 uid=10250 gid=10250}: msd_tool::daemon: Received connection`
 

> ```shell
> /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a client get-functions

`raven:/ # /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a client get-functions
"function0" -> "ffs.adb"`
 
> /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a client set-mass-storage -t disk-ro -f /sdcard/YOUR_IMAGE_FILE.img
> ```

`raven:/ # /data/adb/modules/com.chiller3.msd/msd-tool.arm64-v8a client set-mass-storage -t disk-ro -f /storage/emulated/0/Download/debian-12.7.0-amd64-netinst.iso
Error: Failed to find gadget HAL process`

chenxiaolong, 2024-10-08:
Thanks! That wasn't what I was expecting to fail, but it should be an easy fix once we figure out the root cause.

Can you also post the output of this?

```
ps -efZ | grep -i usb
```

I'm expecting to see something like this:

```
u:r:hal_usb_gadget_impl:s0     system        1070     1 0 17:59:39 ?     00:00:00 android.hardware.usb.gadget-service
```

I'm guessing your device is probably running the USB gadget service under a different SELinux context than `u:r:hal_usb_gadget_impl:s0`.

sethfoxen, 2024-10-09:
> ```
> ps -efZ | grep -i usb
> ```

```
raven:/ # ps -efZ | grep -i usb
u:r:kernel:s0                  root           326     2 0 12:15:22 ?     00:00:00 [wq-tcpm-usb-psy]
u:r:kernel:s0                  root           327     2 0 12:15:22 ?     00:00:00 [wq-tcpm-usb-psy]
u:r:kernel:s0                  root           350     2 0 12:15:22 ?     00:00:00 [google,usbc_por]
u:r:hal_usb_impl:s0            system        1321     1 0 12:15:35 ?     00:02:38 android.hardware.usb-service.gs101
u:r:hal_usb_gadget_impl:s0     system        1322     1 0 12:15:35 ?     00:00:00 android.hardware.usb.gadget-service.gs101
u:r:magisk:s0                  root         11059 10891 2 12:14:55 /debug_ramdisk/.magisk/pts/0 00:00:00 grep -i usb
```

> `dmesg | grep avc`

[dmesg  grep avc.txt](https://github.com/user-attachments/files/17311522/dmesg.grep.avc.txt)

sethfoxen, 2024-10-10:
The latest update has fixed this issue! 
However, I had to disable USB Debugging in my developer options in order to get the emulated image to be recognized by my PC.

chenxiaolong, 2024-10-10:
Awesome, glad it's working for you now!

Did you have the same issue with needing to disable adb when using isodrive? I took a quick look at that project and it looks like it sets things up the same way as MSD.

(I'm guessing it's probably very device dependent. On my Pixel Tablet, adb and mass storage work at the same time, but the USB controller hardware is different from the Pixel 6 Pro.)

sethfoxen, 2024-10-10:
I didn't have to disable adb to use isodrive. For what it's worth, MSD shows that adb is an active gadget function when it's enabled. When it's disabled, it shows that mtp is active instead. 

[image]

chenxiaolong, 2024-10-10:
Hmm, when you use isodrive, what gadget functions are enabled? (You can still use MSD to see.) It it `ffs.adb` and `mass_storage.0`?

sethfoxen, 2024-10-11:
After mounting an image, this is what MSD displays 

[image]

chenxiaolong, 2024-10-11:
Interesting. `ffs.mtp`: so it seems like isodrive is disabling adb for you? I suppose that explains the difference in behavior.

sethfoxen, 2024-10-11:
Oh! I forgot to re-enable debugging haha, it still shows adb active. 

[image]

sethfoxen, 2024-10-11:
Upon more testing, it appears that isodrive ALSO requires me to disable debugging to allow the mounted image to show up on my PC

chenxiaolong, 2024-10-11:
Thanks for testing! Yeah, that's annoying. I don't think anything can be done about that--it's probably a hardware limitation.

sethfoxen, 2024-10-11:
It's alright! The main issue has been taken care of. That's all that matters.

sethfoxen, 2024-10-14:
Oh no! The recent update regressed this issue!

chenxiaolong, 2024-10-15:
I feel like this has to be some sort of installation-related issue. Neither 1.2 -> 1.3 nor 1.3 -> 1.4 changed anything about how MSD looks for the USB gadget HAL process.

Can you please install the test build below, reproduce the problem, and then grab all of `/data/local/tmp/msd/`? (You'll probably have to put them in a zip file to upload to Github.) This build is identical to version 1.4, except with some additional logging.

[MSD-1.4.r2.g3fef462-release.zip](https://github.com/user-attachments/files/17370663/MSD-1.4.r2.g3fef462-release.zip)

sethfoxen, 2024-10-15:
I haven't changed anything major on my phone. Still running the same magisk modules and such. Here are the logs. 

[msd.zip](https://github.com/user-attachments/files/17381229/msd.zip)

sethfoxen, 2024-10-15:
I have quite a lot, actually haha. I'll list them: 
PiF https://github.com/chiteroman/PlayIntegrityFix 
SQLite https://github.com/rojenzaman/sqlite3-magisk-module/releases 
Zygisk Assistant https://github.com/snake-4/Zygisk-Assistant 
Trichrome Library Squoosh https://github.com/entr0pia/trichromelibrary-squoosh 
Unity Fix https://github.com/Magisk-Modules-Alt-Repo/unityfix.git 
Disable Google Analytics https://gitlab.com/adrian.m.miller/disablegoogleanalytics 
Magisk Manager for Recovery https://github.com/Rikj000/Magisk-Manager-for-Recovery-Mode 
Busybox NDK https://github.com/Magisk-Modules-Repo/busybox-ndk 
Pixel Tensor Audio Decompressor https://github.com/JoshuaDoes/pixel-tensor-audio-decompressor 
Safetynet Sensitive Props https://github.com/Magisk-Modules-Alt-Repo/sensitive_props 
Detach https://github.com/Magisk-Modules-Alt-Repo/Detach.git 
Google Photos Unlimited Backup v1 https://www.pling.com/p/2004615/ 
Shamiko https://github.com/LSPosed/LSPosed.github.io 
Simple Bootloop Saver https://github.com/Magisk-Modules-Alt-Repo/Simple_BootloopSaver

chenxiaolong, 2024-10-16:
Haha, you weren't kidding! That's a lot of modules. Thanks for taking the time to list all of those.

I think I found the issue and it doesn't seem to be related to the other modules. The list of running processes now has this:
```
u:r:hal_usb_gadget_impl:s0           1288 system    0:00 /vendor/bin/hw/android.hardware.usb.gadget-service.gs101
```
whereas previously, you had:
```
u:r:hal_usb_gadget_impl:s0     system        1070     1 0 17:59:39 ?     00:00:00 android.hardware.usb.gadget-service
```
Any chance an OS update happened around the same time you updated MSD? That's the only thing that could add the `.gs101` suffix.

I think #16 should fix the problem. It now searches for `android.hardware.usb.gadget-service` as a prefix instead of an exact match. Can you give this test build a try? [MSD-1.4.r4.ge5bd279-release.zip](https://github.com/user-attachments/files/17389221/MSD-1.4.r4.ge5bd279-release.zip)

If it still doesn't work, please grab a new set of logs.

sethfoxen, 2024-10-16:
This has fixed the issue once more! 
There WAS an OTA update on the 14th, and a quick glance suggests they did SOMETHING with the hal process. (https://download.lineageos.org/devices/raven/changes) 
As an aside, with this build I'm able to keep adb enabled, and even access MTP while having an image mounted.

chenxiaolong, 2024-10-16:
Awesome, thanks for testing! I'll release MSD 1.5 after work tonight with these changes included.

> As an aside, with this build I'm able to keep adb enabled, and even access MTP while having an image mounted.

Sounds like LineageOS (or Google) made some nice improvements with this update, even if it temporarily impacted MSD :slightly_smiling_face:

commit e5bd279a (2024-10-17): Search for the gadget HAL using a prefix instead of the full name

On the Pixel 6 Pro, it seems that the executable has an extra `.gs101`
suffix.

Fixes: #12
PR #16 (2024-10-17): Search for the gadget HAL using a prefix instead of the full name
On the Pixel 6 Pro, it seems that the executable has an extra `.gs101` suffix.

Fixes: #12
closed 2024-10-17 by chenxiaolong