chenxiaolong/MSD #17: "Permission denied" errors in MSD app (cli tool works fine)

fnr1r, 2024-10-18:
Hello.
I have an issue similar to #11 and #12 (that being `Failed to <issue here>` with `IOException (Permission denied)`) but on a Pixel 8 Pro (Android 14, stock rom) and with errors in dmesg. The cli tool works fine.

[msd.post-fs-data.log](https://github.com/user-attachments/files/17436386/msd.post-fs-data.log)
[msd.service.log](https://github.com/user-attachments/files/17436392/msd.service.log)
[policy](https://files.catbox.moe/6wkeoc)

```
┌root@localhost:/
└─# dmesg | grep msd | grep denied                                                        
[16269.514196] type=1400 audit(1729257752.805:11531): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16272.757797] type=1400 audit(1729257756.049:11554): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16273.884507] type=1400 audit(1729257757.177:11556): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16275.185462] type=1400 audit(1729257758.477:11559): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16275.193905] type=1400 audit(1729257758.481:11560): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16496.345762] type=1400 audit(1729257979.633:11672): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16497.285839] type=1400 audit(1729257980.569:11673): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16497.300129] type=1400 audit(1729257980.593:11674): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16498.216146] type=1400 audit(1729257981.509:11675): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16498.223125] type=1400 audit(1729257981.517:11676): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16791.693498] type=1400 audit(1729258274.985:11718): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16793.088436] type=1400 audit(1729258276.381:11719): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16793.097740] type=1400 audit(1729258276.385:11720): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16793.723194] type=1400 audit(1729258277.017:11721): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16793.729322] type=1400 audit(1729258277.021:11722): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16841.817557] type=1400 audit(1729258325.109:11724): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16848.213804] type=1400 audit(1729258331.505:11726): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16848.766536] type=1400 audit(1729258332.057:11727): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16849.263557] type=1400 audit(1729258332.557:11728): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16849.470926] type=1400 audit(1729258332.765:11729): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16849.723427] type=1400 audit(1729258333.017:11730): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16849.731141] type=1400 audit(1729258333.025:11731): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16849.904650] type=1400 audit(1729258333.197:11732): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16967.898464] type=1400 audit(1729258451.193:11767): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[16984.374142] type=1400 audit(1729258467.665:11768): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0 app=com.chiller3.msd
[17173.390835] type=1400 audit(1729258656.685:11789): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17175.703623] type=1400 audit(1729258658.997:11791): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17176.344092] type=1400 audit(1729258659.637:11792): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17176.741462] type=1400 audit(1729258660.033:11793): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17574.890718] type=1400 audit(1729259058.177:11822): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17578.193879] type=1400 audit(1729259061.485:11823): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17579.318580] type=1400 audit(1729259062.613:11824): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[... cut ...] :s0 tclass=unix_stream_socket permissive=0
[17580.262661] type=1400 audit(1729259063.557:11826): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
[17606.680201] type=1400 audit(1729259089.973:11828): avc:  denied  { connectto } for  comm="DefaultDispatch" path=006D736464 scontext=u:r:priv_app:s0:c512,c768 tcontext=u:r:msd_daemon:s0 tclass=unix_stream_socket permissive=0
```

chenxiaolong, 2024-10-18:
Can you try upgrading to MSD 1.5? This looks like something that should have been fixed in versions >=1.3.

If you still encounter issues on that version, please upload everything from `/data/local/tmp/msd/`. Thanks!

fnr1r, 2024-10-18:
I am on v1.5.

[msd.zip](https://github.com/user-attachments/files/17437382/msd.zip)

fnr1r, 2024-10-18:
I have:

- MSD
- "pm install" from anywhere
- Systemless Mkshrc
- MMRL CLI
- WiFi Password Viewer for MMRL
- Nano for Android NDK
- Pixel Xpert
- Play Integrity Fix
- SSH for Magisk
- systemless hosts
- Tricky Store
- TSupport
- UnityFix
- Zygisk - LSPosed (Jing Matrix fork)
- Zygisk - Sui
- Zygisk Next

Please note that I'm using KernelSU and I have the "Umount modules by default" option enabled.

As for `/system/etc/selinux/plat_seapp_contexts`:

```txt
isSystemServer=true domain=system_server_startup
user=_app seinfo=platform name=com.android.traceur domain=traceur_app type=app_data_file levelFrom=all
user=system seinfo=platform domain=system_app type=system_app_data_file
user=system seinfo=platform isPrivApp=true name=com.android.DeviceAsWebcam domain=device_as_webcam type=system_app_data_file levelFrom=all
user=bluetooth seinfo=bluetooth domain=bluetooth type=bluetooth_data_file
user=network_stack seinfo=network_stack domain=network_stack type=radio_data_file
user=nfc seinfo=platform domain=nfc type=nfc_data_file
user=nfc seinfo=nfc domain=nfc type=nfc_data_file
user=secure_element seinfo=platform domain=secure_element levelFrom=all
user=radio seinfo=platform domain=radio type=radio_data_file
user=shared_relro domain=shared_relro levelFrom=all
user=shell seinfo=platform domain=shell name=com.android.shell type=shell_data_file
user=webview_zygote seinfo=webview_zygote domain=webview_zygote
user=_isolated domain=isolated_app levelFrom=user
user=_isolated isIsolatedComputeApp=true domain=isolated_compute_app levelFrom=user
user=_sdksandbox domain=sdk_sandbox_34 type=sdk_sandbox_data_file levelFrom=all
user=_sdksandbox isSdkSandboxNext=true domain=sdk_sandbox_next type=sdk_sandbox_data_file levelFrom=all
user=_sdksandbox isSdkSandboxAudit=true domain=sdk_sandbox_audit type=sdk_sandbox_data_file levelFrom=all
user=_app seinfo=app_zygote domain=app_zygote levelFrom=user
user=_app seinfo=media domain=mediaprovider type=app_data_file levelFrom=user
user=_app seinfo=platform domain=platform_app type=app_data_file levelFrom=user
user=_app isEphemeralApp=true domain=ephemeral_app type=app_data_file levelFrom=all
user=_app isPrivApp=true domain=priv_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.permissioncontroller domain=permissioncontroller_app type=privapp_data_file levelFrom=all
user=_app seinfo=media isPrivApp=true name=com.android.providers.media.module domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app seinfo=media isPrivApp=true name=com.android.providers.media.module:* domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.providers.media.module domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.providers.media.module:* domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app seinfo=platform isPrivApp=true name=com.android.permissioncontroller domain=permissioncontroller_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.android.vzwomatrigger domain=vzwomatrigger_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.android.rkpdapp domain=rkpdapp type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.rkpdapp domain=rkpdapp type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.gms domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.gms.* domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.gms:* domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.gsf domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app minTargetSdkVersion=34 domain=untrusted_app type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=32 domain=untrusted_app_32 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=30 domain=untrusted_app_30 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=29 domain=untrusted_app_29 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=28 domain=untrusted_app_27 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=26 domain=untrusted_app_27 type=app_data_file levelFrom=user
user=_app domain=untrusted_app_25 type=app_data_file levelFrom=user
user=_app minTargetSdkVersion=28 fromRunAs=true domain=runas_app levelFrom=all
user=_app fromRunAs=true domain=runas_app levelFrom=user
```

And a diff:

```
--- /system/etc/selinux/plat_seapp_contexts     2009-01-01 01:00:00.000000000 +0100
+++ com.chiller3.msd/seapp_temp/plat_seapp_contexts     2024-10-18 17:41:55.556000001 +0200
@@ -43,3 +43,4 @@
 user=_app domain=untrusted_app_25 type=app_data_file levelFrom=user
 user=_app minTargetSdkVersion=28 fromRunAs=true domain=runas_app levelFrom=all
 user=_app fromRunAs=true domain=runas_app levelFrom=user
+user=_app isPrivApp=true name=com.chiller3.msd domain=msd_app type=app_data_file levelFrom=all
```

chenxiaolong, 2024-10-18:
Thanks for the info! I'll need to see if I can get KernelSU to run in an emulator. I don't have a device I can test that on at the moment.

fnr1r, 2024-10-20:
I got it working by copying `/data/adb/modules/com.chiller3.msd/seapp_temp/plat_seapp_contexts` to `/data/adb/modules/com.chiller3.msd/system/etc/selinux/plat_seapp_contexts`. It works since it's included in the overlay mount.

`mount` output:
```
/dev/block/loop48 on /data/adb/modules type ext4 (rw,seclabel,relatime)
KSU on /debug_ramdisk type tmpfs (rw,seclabel,relatime)
com.chiller3.msd on /data/adb/modules/com.chiller3.msd/seapp_temp type tmpfs (rw,seclabel,relatime)
com.chiller3.msd on /system/etc/selinux/plat_seapp_contexts type tmpfs (rw,seclabel,relatime)
KSU on /system type overlay (ro,seclabel,relatime,lowerdir=/data/adb/modules/PixelXpert/system:/data/adb/modules/mmrl_wpd/system:/data/adb/modules/nano-ndk/system:/data/adb/modules/mkshrc/system:/data/adb/modules/com.chiller3.msd/system:/data/adb/modules/mmrl/system:/data/adb/modules/ssh/system:/system)
KSU on /vendor type overlay (ro,seclabel,relatime,lowerdir=/data/adb/modules/mkshrc/vendor:/vendor)
KSU on /system_ext type overlay (ro,seclabel,relatime,lowerdir=/data/adb/modules/mkshrc/system_ext:/system_ext)
KSU on /product type overlay (ro,seclabel,relatime,lowerdir=/data/adb/modules/PixelXpert/product:/product)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
mtp on /dev/usb-ffs/mtp type functionfs (rw,relatime)
ptp on /dev/usb-ffs/ptp type functionfs (rw,relatime)
KSU on /system/etc type overlay (rw,seclabel,relatime,lowerdir=/system/etc,upperdir=/data/adb/modules/systemless-hosts-KernelSU-module/system/etc,workdir=/data/adb/modules/systemless-hosts-KernelSU-module/worker)
```

chenxiaolong, 2024-10-20:
```
com.chiller3.msd on /system/etc/selinux/plat_seapp_contexts type tmpfs (rw,seclabel,relatime)
KSU on /system type overlay (ro,seclabel,relatime,lowerdir=/data/adb/modules/PixelXpert/system:/data/adb/modules/mmrl_wpd/system:/data/adb/modules/nano-ndk/system:/data/adb/modules/mkshrc/system:/data/adb/modules/com.chiller3.msd/system:/data/adb/modules/mmrl/system:/data/adb/modules/ssh/system:/system)
```

Ah, that explains it. So KernelSU uses overlayfs. overlayfs' `lowerdir` option does not "follow" other things mounted underneath `/system`, so it becomes invisible.

The reason MSD doesn't use `/data/adb/modules/com.chiller3.msd/system/` is because both Magisk and KernelSU wait for all modules to be loaded before mounting files from any of them.

When multiple modules need to modify the same file, this causes an issue because they'll each read the original file from the ROM and add their own line. Each one will never see the modification from another module because Magisk/KernelSU's mounting happens too late. The last loaded module ends up "winning" and the other ones just break.

fnr1r, 2024-10-21:
Yep. This behaviour matches ksud code.

[on_post_data_fs (line 99)](https://github.com/tiann/KernelSU/blob/59ba84eb2b45c7ee7cded8774c918100d62a2712/userspace/ksud/src/init_event.rs#L99)
[post-fs-data script runner (line 200)](https://github.com/tiann/KernelSU/blob/59ba84eb2b45c7ee7cded8774c918100d62a2712/userspace/ksud/src/init_event.rs#L200)
[overlay mounted (line 210)](https://github.com/tiann/KernelSU/blob/59ba84eb2b45c7ee7cded8774c918100d62a2712/userspace/ksud/src/init_event.rs#L210)

chenxiaolong, 2024-10-21:
Thanks for linking to that. Looks like KernelSU supports an [additional `post-mount.sh` hook](https://github.com/tiann/KernelSU/blob/59ba84eb2b45c7ee7cded8774c918100d62a2712/userspace/ksud/src/init_event.rs#L214) where MSD could do the mounting.

Just need to find a reliable way to detect Magisk vs. KernelSU to determine if the mounting should happen in `post-fs-data.sh` or `post-mount.sh`. It seems like checking for `KSU=true` [should be sufficient](https://github.com/tiann/KernelSU/blob/7b3e732404618fc3ee8940c20d1bd621cf57d332/userspace/ksud/src/module.rs#L180). I'll try that after work.

fnr1r, 2024-10-21:
[That's how HuskyDG does it.](https://github.com/HuskyDG/magic_overlayfs/blob/e5e6087d206f4fa4ed24683484c3df93eb258c27/magisk-module/customize.sh#L130)
Also I have a fix ready. I'll make a PR in a second.

fnr1r, 2024-10-21:
I changed my approach to fixing this issue. Instead of mounting the modified file, I copy it to the module's `system` directory and copy the a/m time. Since PR #18 is broken for people who use the systemless hosts module, I'd rather close that one and make a new one with the `kernelsu-fix2` branch instead.

chenxiaolong, 2024-10-21:
Darn, that's annoying that the systemless hosts module mounts the entire `/system/etc`.

I can't merge the `kernelsu-fix2` fix though because it's not compatible with having two modules both edit `plat_seapp_contexts`. (Another one of my own modules modifies that file too.)

fnr1r, 2024-10-22:
Maybe patching `plat_seapp_contexts` should be a separate module then (just like Zygisk). Having two modules patch and mount a file is kinda janky, but having one module that does that based on info from other modules would be better.

PR #18 (2024-10-26): Fix SELinux policy patching on KernelSU
Fixes #17.