chenxiaolong/MSD #19: Permission denied (os error 13)

ukrbugreporter, 2024-11-01:
I am using Poco X3 Pro (vayu) running crDroid 10.9 based on Android 14 rooted with Kitsune.
Configfs, USB Gadget and USB gadget HAL seem to be supported:

```
vayu:/ # ls /config/usb_gadget/g1
UDC              bDeviceSubClass  bcdUSB     idProduct  strings
bDeviceClass     bMaxPacketSize0  configs    idVendor
bDeviceProtocol  bcdDevice        functions  os_desc
vayu:/ # zcat /proc/config.gz | grep CONFIG_USB_CONFIGFS_MASS_STORAGE
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
vayu:/ # pgrep -f android.hardware.usb.gadget-service
1|vayu:/ # 
```

I followed the installation instructions but after booting into system and launching MSD I get an error. It is either
```
Failed to open directory: "/config/usb_gadget/g1/configs/b.1"

Caused by:
    Permission denied (os error 13)
```

or

`Failed to find gadget HAL process: android.hardware.usb.gadget-service*`

The content of /data/local/tmp/msd/msd-tool.log file looks like this:

```
--------- beginning of main
11-01 15:46:30.644  1699  5254 I msd-tool: peer{pid=5174 uid=10477 gid=10477}: msd_tool::daemon: Received connection
11-01 15:46:30.644  1699  5254 W msd-tool: peer{pid=5174 uid=10477 gid=10477}: msd_tool::daemon: Failed to open directory: "/config/usb_gadget/g1/configs/b.1"
11-01 15:46:30.644  1699  5254 W msd-tool: 
11-01 15:46:30.644  1699  5254 W msd-tool: Caused by:
11-01 15:46:30.644  1699  5254 W msd-tool:     Permission denied (os error 13)
11-01 15:50:22.907  1699  6932 I msd-tool: peer{pid=6078 uid=10477 gid=10477}: msd_tool::daemon: Received connection
11-01 15:50:22.907  1699  6932 W msd-tool: peer{pid=6078 uid=10477 gid=10477}: msd_tool::daemon: Failed to open directory: "/config/usb_gadget/g1/configs/b.1"
11-01 15:50:22.907  1699  6932 W msd-tool: 
11-01 15:50:22.907  1699  6932 W msd-tool: Caused by:
11-01 15:50:22.907  1699  6932 W msd-tool:     Permission denied (os error 13)
```

I checked if the directory /config/usb_gadget/g1/configs/b.1/ actually exists and it does

```
vayu:/ # ls /config/usb_gadget/g1/configs/b.1/                                    
MaxPower  bmAttributes  f1  strings
```

Something weird happens on my Sony Xperia XA2 Plus too. It is running LineageOS 21 based on Android 14, again, rooted with Kitsune.

Support for MSD seems to be present:

```
H4413:/ # ls /config/usb_gadget/g1
UDC              bDeviceSubClass  bcdUSB     idProduct  strings
bDeviceClass     bMaxPacketSize0  configs    idVendor
bDeviceProtocol  bcdDevice        functions  os_desc
H4413:/ # zcat /proc/config.gz | grep CONFIG_USB_CONFIGFS_MASS_STORAGE
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
H4413:/ # pgrep -f android.hardware.usb.gadget-service
1|H4413:/ #
```

But the app does not even start. It hangs and then crashes. /data/local/tmp/msd/msd-tool.log is empty, /sdcard/Android/com.chiller3.msd/ is non-existent.

ukrbugreporter, 2024-11-01:
Just checked other issues and I did the `adb shell su -c 'dmesg | grep denied' > denials.txt`

[Output](https://0x0.st/XGAp.txt)

chenxiaolong, 2024-11-01:
Can you zip all of `/data/local/tmp/msd/` and upload that? I'll take a look after work today.

Also, can you post the output of:

```
adb shell su -c 'ls -lZ /config/usb_gadget/g1'
```

> But the app does not even start. It hangs and then crashes. /data/local/tmp/msd/msd-tool.log is empty, /sdcard/Android/com.chiller3.msd/ is non-existent.

Between the two devices, are there any difference on configuration for root hiding modules? This sounds very much like Android itself can't access MSD's apk file.

> ```
> H4413:/ # pgrep -f android.hardware.usb.gadget-service
> 1|H4413:/ #
> ```

This didn't return any output, so both your devices aren't using the gadget HAL. MSD currently doesn't support this.

The reason MSD talks to the gadget HAL at all is to prevent Android from messing with the USB state while MSD is making its own changes. This might not be necessary on devices that used the old USB configuration mechanism from before Google introduced the gadget HAL.

But let's figure out the permissions issues first.

ukrbugreporter, 2024-11-02:
I have uninstalled MSD since yesterday and after reinstalling the app does not start on Poco at all, same as Sony. I did get a [crash log](https://paste.crdroid.net/NLSodN) however.

> Can you zip all of `/data/local/tmp/msd/` and upload that?

[msd.zip](https://github.com/user-attachments/files/17606978/msd.zip)

> adb shell su -c 'ls -lZ /config/usb_gadget/g1'

```
total 0
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 UDC
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 bDeviceClass
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 bDeviceProtocol
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 bDeviceSubClass
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 bMaxPacketSize0
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 bcdDevice
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 bcdUSB
drwxr-xr-x  3 root root u:object_r:configfs:s0     0 2024-11-02 12:23 configs
drwxr-xr-x 12 root root u:object_r:configfs:s0     0 2024-11-02 12:23 functions
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 idProduct
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-11-02 12:23 idVendor
drwxr-xr-x  2 root root u:object_r:configfs:s0     0 2024-11-02 12:23 os_desc
drwxr-xr-x  3 root root u:object_r:configfs:s0     0 2024-11-02 12:23 strings
```

> Between the two devices, are there any difference on configuration for root hiding modules? This sounds very much like Android itself can't access MSD's apk file.

Both devices are rooted with Kitsune with Zygisk, MagiskHide and SuList enabled. At the time of opening the issue I had [playcurlNEXT](https://paste.crdroid.net/NLSodN) installed on Poco, but I have uninstalled it since.

chenxiaolong, 2024-11-03:
Thanks for the logs. I installed Kitsune and was able to reproduce the problem. It looks like even though the app part of MSD doesn't use root, you still have to add it to SuList in Kitsune. That fixes the crashes on my device.

Please give that a try. If things are working, then I'd expect it to fail with `Failed to find gadget HAL process: android.hardware.usb.gadget-service*` every time.

If you're able to get to that point, then please try this test build (which no longer fails if the gadget HAL is missing): [MSD-1.5.r5.ge8bb80e-release.zip](https://github.com/user-attachments/files/17608685/MSD-1.5.r5.ge8bb80e-release.zip)

ukrbugreporter, 2024-11-03:
Thanks for reply!

> you still have to add it to SuList in Kitsune

Totally forgot that I did than on first installation. Now the app doesn't crash.

> Please give that a try. If things are working, then I'd expect it to fail with `Failed to find gadget HAL process: android.hardware.usb.gadget-service*` every time.

Yes, this is the case.

Anyway, in a new build after tapping "Activate USB gadget functions" or "Apply settings" I face these errors:

```
Failed to write file: "/config/usb_gadget/g1/UDC"

Caused by:
    Permission denied (os error 13)
```
```
Failed to write file: "/config/usb_gadget/g1/b.1"

Caused by:
    Permission denied (os error 13)
```

> That fixes the crashes on my device

But did it make the module work with Kitsune?

chenxiaolong, 2024-11-03:
No problem!

Yep, it's working on my device with Kitsune.

I just noticed in your `adb shell su -c 'ls -lZ /config/usb_gadget/g1'` output that all the files are owned by `root` instead of `system`. That unusual to me, but might be normal for devices without the gadget HAL. I'll see if I can come up with a workaround tomorrow.

As a quick test, if you run:
```
adb shell su -c "chown -R system:system /config/usb_gadget/g1"
```
does MSD start working?

ukrbugreporter, 2024-11-03:
Yes, thank you! Hope you'll merge the changes soon.

Yet I am still not sure why I can't boot with my phone. It is detected by UEFI, but even if I put it first in the boot order it will be skipped. The same happens with Isodrive. I've read somewhere that it has to do with images not supporting UEFI if that makes any sense but again I am not sure.

ukrbugreporter, 2024-11-03:
Nevermind, just tested with Arch Linux ISO and it works. Must be a lack of legacy boot support in my UEFI.

chenxiaolong, 2024-11-03:
No problem!

Please give this test build a try: [MSD-1.5.r5.g9aa2206-release.zip](https://github.com/user-attachments/files/17612472/MSD-1.5.r5.g9aa2206-release.zip). It should no longer require manually running `chown`.

> I've read somewhere that it has to do with images not supporting UEFI if that makes any sense but again I am not sure.
>
> Nevermind, just tested with Arch Linux ISO and it works. Must be a lack of legacy boot support in my UEFI.

Yeah, most new computers no longer support legacy boot at all. For compatibility, using a read-only disk in MSD/isodrive usually works much better than a CD-ROM.

closed 2024-11-08 by chenxiaolong

Anarchtism, 2025-01-01:
I am on the latest 1.9 release and I still have to run `su -c "chown -R system:system /config/usb_gadget/g1"` to get around
```
Failed to read directory: "/config/usb_gadget/g1/configs/b.1"

Caused by:
    Permission denied (os error 13)
```

chenxiaolong, 2025-01-01:
@Anarchtism Can you create a new issue for that and attach a zip of everything in `/data/local/tmp/msd/`?

commit 6347b275 (2024-11-03): msd-tool: Add support for devices without gadget HAL

Older devices use init scripts for managing the USB controller and they
don't have any process running that constantly tries to ensure that the
USB controller state is correct.

On these devices, we can skip sending SIGSTOP/SIGCONT, but we might need
to chown the configfs directory since it could be owned by root.

Fixes: #19
commit 9aa2206b (2024-11-03): msd-tool: Add support for devices without gadget HAL

Older devices use init scripts for managing the USB controller and they
don't have any process running that constantly tries to ensure that the
USB controller state is correct.

On these devices, we can skip sending SIGSTOP/SIGCONT, but we might need
to chown the configfs directory since it could be owned by root.

Fixes: #19
PR #20 (2024-11-08): msd-tool: Add support for devices without gadget HAL
Older devices use init scripts for managing the USB controller and they don't have any process running that constantly tries to ensure that the USB controller state is correct.

On these devices, we can skip sending SIGSTOP/SIGCONT, but we might need to chown the configfs directory since it could be owned by root.

Fixes: #19