chenxiaolong/MSD #21: IOException (Permission denied) | KernelSU

Damglador, 2024-11-11:
IOException (Permission denied) when opening the app.
[image]

Downloaded from GitHub, installed in KernelSU, rebooted.

[image]

When I found this module the same thing happened, on version 1.5, so I checked GitHub issues, saw that that's a KernelSU issue and decided to wait for 1.6 to test it again.

I have other module, but it's CLI, called isodrive and it works, with some quiks, but mostly fine.

Device: Ulefone Armor 21
OS: OEM Android 14
Root: KernelSU 11928 LKM

chenxiaolong, 2024-11-11:
Do you have the systemless hosts module installed? That one is currently known to conflict with MSD.

If not, can you zip up the files in `/data/local/tmp/msd` and upload that?

Damglador, 2024-11-11:
Damn, I do, for AdAway. I'll check if it works without it

Damglador, 2024-11-11:
> Do you have the systemless hosts module installed? That one is currently known to conflict with MSD.
> 
> If not, can you zip up the files in `/data/local/tmp/msd` and upload that?

Disables systemless hosts and now error is ``Failed to open directory: "/config/usb_gadget/g1/configs/b.1"

Caused by:
    Permission denied (os error 13)``

[image]

chenxiaolong, 2024-11-11:
Hmm, Can you post the output of:
```
adb shell su -c 'ls -lZR /config/usb_gadget'
```
after reproducing the error? This should have been fixed in 1.6.

If you could post a link to the ROM you're running, that'd be helpful too.

Damglador, 2024-11-12:
```
~ $ sudo ls -lZR /config/usb_gadget
/config/usb_gadget:
total 0
drwxrwx---. 6 shell shell u:object_r:configfs:s0 0 Nov 12 04:56 g1

/config/usb_gadget/g1:
total 0
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 UDC
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 bDeviceClass
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 bDeviceProtocol
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 bDeviceSubClass
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:58 bMaxPacketSize0
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 bcdDevice
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 bcdUSB
drwxr-xr-x.  3 root root u:object_r:configfs:s0    0 Nov 11 21:04 configs
drwxr-xr-x. 17 root root u:object_r:configfs:s0    0 Nov 11 21:04 functions
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 idProduct
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 idVendor
-rw-r--r--.  1 root root u:object_r:configfs:s0 4096 Nov 12 04:58 max_speed
drwxr-xr-x.  2 root root u:object_r:configfs:s0    0 Nov 12 04:56 os_desc
drwxr-xr-x.  3 root root u:object_r:configfs:s0    0 Nov 11 21:04 strings

/config/usb_gadget/g1/configs:
total 0
drwxrwx---. 3 shell shell u:object_r:configfs:s0 0 Nov 12 04:56 b.1

/config/usb_gadget/g1/configs/b.1:
total 0
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 MaxPower
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 12 04:58 bmAttributes
lrwxrwxrwx. 1 root root u:object_r:configfs:s0    0 Nov 12 04:54 f1 -> ../../../../usb_gadget/g1/functions/ffs.mtp
lrwxrwxrwx. 1 root root u:object_r:configfs:s0    0 Nov 12 04:54 f2 -> ../../../../usb_gadget/g1/functions/ffs.adb
drwxr-xr-x. 3 root root u:object_r:configfs:s0    0 Nov 11 21:04 strings

/config/usb_gadget/g1/configs/b.1/strings:
total 0
drwxrwx---. 2 shell shell u:object_r:configfs:s0 0 Nov 12 04:56 0x409

/config/usb_gadget/g1/configs/b.1/strings/0x409:
total 0
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 configuration

/config/usb_gadget/g1/functions:
total 0
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 accessory.gs2
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 acm.gs0
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 acm.gs1
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 acm.gs2
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 acm.gs3
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 audio_source.gs3
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 ffs.adb
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 ffs.mtp
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 ffs.ptp
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 hid.gs0
drwxr-xr-x. 3 root root u:object_r:configfs:s0 0 Nov 12 04:56 mass_storage.usb0
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 midi.gs5
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 via_atc.gs0
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 via_ets.gs0
drwxr-xr-x. 2 root root u:object_r:configfs:s0 0 Nov 11 21:04 via_modem.gs0

/config/usb_gadget/g1/functions/accessory.gs2:
total 0

/config/usb_gadget/g1/functions/acm.gs0:
total 0
-r--r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 12 04:58 port_num

/config/usb_gadget/g1/functions/acm.gs1:
total 0
[... 59 lines cut ...]

/config/usb_gadget/g1/functions/via_modem.gs0:
total 0

/config/usb_gadget/g1/os_desc:
total 0
lrwxrwxrwx. 1 root root u:object_r:configfs:s0    0 Nov 11 21:04 b.1 -> ../../../usb_gadget/g1/configs/b.1
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 b_vendor_code
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 qw_sign
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 12 04:54 use

/config/usb_gadget/g1/strings:
total 0
drwxrwx---. 2 root root u:object_r:configfs:s0 0 Nov 12 04:56 0x409

/config/usb_gadget/g1/strings/0x409:
total 0
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 manufacturer
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 product
-rw-r--r--. 1 root root u:object_r:configfs:s0 4096 Nov 11 21:04 serialnumber
```

Fresh from manufacturer https://mega.nz/folder/F24WmAyZ#gzRxjLgjk0Y1YuDtkh_1Dg, even sent me flash software :D

boot.img patched with KernelSU app and flashed with `fastboot boot flash /path/boot.img` (if I remember correctly)

chenxiaolong, 2024-11-12:
Thanks! This is the issue:
```
/config/usb_gadget:
total 0
drwxrwx---. 6 shell shell u:object_r:configfs:s0 0 Nov 12 04:56 g1
```

On newer devices that use Android's USB gadget HAL, these files are owned by the `system` user. Older devices from before Google introduced the USB gadget HAL had these files owned by `root`. MSD will try and fix the permissions if it sees that `/config/usb_gadget/g1` is owned by `root`, but it didn't do this because the directory is owned by `shell` on your device.

This should be an easy fix. I'll have a test build shortly.

chenxiaolong, 2024-11-12:
Please give #23 a try. Test build: [MSD-1.6.r4.gde76ef3-release.zip](https://github.com/user-attachments/files/17711482/MSD-1.6.r4.gde76ef3-release.zip)

If it still doesn't work, please upload the new logs from `/data/local/tmp/msd`.

Damglador, 2024-11-12:
```
Failed to open directory: "/config/usb_gadget/g1/."

Caused by:
    Permission denied (os error 13)
```
[image]

chenxiaolong, 2024-11-12:
Thanks! I'll take a look after work today.

chenxiaolong, 2024-11-12:
Please give this a try: [MSD-1.6.r4.ga8c6ef7-release.zip](https://github.com/user-attachments/files/17723915/MSD-1.6.r4.ga8c6ef7-release.zip)

Vagelis1608, 2024-11-13:
I also had "os error 13" on v1.6.
It seems to work on r4, it's too late to do a functionality test, but no errors. 

(Nothing Phone 2, Stock Rom v3.0 Open Beta, Android 15)

Damglador, 2024-11-13:
Tested with Vanilla OS image, it boots

chenxiaolong, 2024-11-13:
Great, thanks for testing! I'll release version 1.7 with the fix.

commit 10bb07bc (2024-11-12): usb: Use faccessat() to determine if chown is necessary

ROMs that use AOSP's default /system/etc/hw/init.rc, but without using
the USB gadget HAL will have the USB gadget configfs root owned by
shell:shell.

Fixes: #21
commit de76ef3c (2024-11-12): usb: Use faccessat() to determine if chown is necessary

ROMs that use AOSP's default /system/etc/hw/init.rc, but without using
the USB gadget HAL, will have the USB gadget configfs root owned by
shell:shell.

Fixes: #21
commit a8c6ef7a (2024-11-12): usb: Use faccessat() to determine if chown is necessary

ROMs that use AOSP's default /system/etc/hw/init.rc, but without using
the USB gadget HAL, will have the USB gadget configfs root owned by
shell:shell with 770 permissions.

Fixes: #21
PR #23 (2024-11-13): usb: Use faccessat() to determine if chown is necessary
ROMs that use AOSP's default `/system/etc/hw/init.rc`, but without using the USB gadget HAL, will have the USB gadget configfs root owned by `shell:shell`.

Fixes: #21
closed 2024-11-13 by chenxiaolong