chenxiaolong/MSD #28: [FR] Renaming devices

Damglador, 2024-11-15:
Would be nice to be able to give custom names to devices to keep track of which one is which. Also having path to the local file of a device.

chenxiaolong, 2024-11-15:
I'm not quite sure I understand what you mean. MSD already shows the local path in the UI.

Damglador, 2024-11-16:
Only "Writable media" and semi cryptic path. But I encountered a bigger issue after an update, so this one is for later xd

chenxiaolong, 2024-12-31:
> Yep, can't tell which one I need, so I always turn on both🤣Screenshot_20241231-211815.png

Do you remember how exactly you picked those files in the file selector? Eg. was it from the recent files section, the Downloads folder (via top-level hamburger menu option), Downloads folder (via the "Internal Storage" hamburger menu option), etc.?

chenxiaolong, 2025-01-01:
Thanks for testing that! I'm able to reproduce the issue.

With #34, MSD now shows the underlying filesystem path instead of the URI provided by Android. This should be consistent in every scenario.

[Screenshot_20250101-000938](https://github.com/user-attachments/assets/4d8bc7cc-90c4-4580-b1bd-aaac39fffaaf)

commit 445cac34 (2025-01-01): Show local path instead of URI where possible

Not every document provider includes a path in the URI, which makes it
impossible to distinguish between multiple devices. Since mass storage
device emulation can only work with local files, we can try to show the
local path instead. If the file is no longer accessible (eg. due to it
being deleted), the UI will fall back to the current behavior of showing
the URI.

Fixes: #28
PR #34 (2025-01-01): Show local path instead of URI where possible
Not every document provider includes a path in the URI, which makes it impossible to distinguish between multiple devices. Since mass storage device emulation can only work with local files, we can try to show the local path instead. If the file is no longer accessible (eg. due to it being deleted), the UI will fall back to the current behavior of showing the URI.

Fixes: #28
closed 2025-01-01 by chenxiaolong