chenxiaolong/MSD #29: After updating to v1.8 app doesn't launch

Damglador, 2024-11-16:
Well that's pretty much it. I had this test version for KernelSU, 1.6, noticed that 1.8 released, updated and the GUI app doesn't launch, says "Stopped working" or something and thats it. Tried removing the module, rebooting, installing it back, rebooting and launching the app, but still the same issue.

chenxiaolong, 2024-11-16:
Can you grab a logcat right after it crashes and upload that?
```
adb logcat -d > logcat.txt
```

Damglador, 2024-11-16:
[logcat.txt](https://github.com/user-attachments/files/17787278/logcat.txt)

chenxiaolong, 2024-11-16:
Can you try that again with `adb` or if you want to use Termux, run it as root?

When `logcat` is run from Termux without `su`, it only shows Termux's own logs.

Damglador, 2024-11-16:
Oopsie, sorry for that. 

The crash might be not right at the end
[image]

chenxiaolong, 2024-11-17:
Thanks! This is the issue:
```
11-16 22:08:34.550  2173  2235 E ndroid.systemui: Failed to open APK '/system/priv-app/com.chiller3.msd/app-release.apk': I/O error
```

It looks like Android can't access the APK file at all. Since you already tried uninstalling and installing again, I have no idea what could possibly cause that. I've only ever seen `I/O error` when the filesystem was corrupted.

asislife20, 2025-09-25:
Had the same issue, and ended up having to grant su access in ksu. Unfortunately now I get IOException (Failed to connect to MSD daemon) -> IOException (Connection reset by peer).