chenxiaolong/MSD #32: Connect reset by peer

debianroot20200524, 2024-11-23:
[image]

kali:/data/adb/modules/com.chiller3.msd # ./msd-tool.arm64-v8a help
Usage: msd-tool.arm64-v8a [OPTIONS] <COMMAND>

Commands:
  client   Send messages to daemon
  daemon   Run daemon
  sepatch  Patch SELinux policy file
  help     Print this message or the help of the given subcommand(s)

Options:
  -h, --help  Print help

Global options:
      --log-target <TARGET>  Where to output log messages [default: stderr] [possible values: stderr, logcat]
      --log-level <LEVEL>    Lowest log message severity to output [default: INFO]
kali:/data/adb/modules/com.chiller3.msd # ./msd-tool.arm64-v8a daemon
2024-11-23T05:32:40.027254Z ERROR msd_tool: Failed to listen on domain socket

Caused by:
    Address already in use (os error 98)
1|kali:/data/adb/modules/com.chiller3.msd # ./msd-tool.arm64-v8a client help
Send messages to daemon

Usage: msd-tool.arm64-v8a client [OPTIONS] <COMMAND>

Commands:
  get-functions     Get currently active USB controller functions
  set-mass-storage  Set USB controller to emulate mass storage devices
  get-mass-storage  Get currently active mass storage devices
  help              Print this message or the help of the given subcommand(s)

Options:
  -h, --help  Print help

Global options:
      --log-target <TARGET>  Where to output log messages [default: stderr] [possible values: stderr, logcat]
      --log-level <LEVEL>    Lowest log message severity to output [default: INFO]
kali:/data/adb/modules/com.chiller3.msd # ./msd-tool.arm64-v8a client --log-level DEBUG get-functions
2024-11-23T05:33:47.870524Z ERROR msd_tool: Failed to receive protocol version acknowledgement

Caused by:
    Connection reset by peer (os error 104)
1|kali:/data/adb/modules/com.chiller3.msd # ./msd-tool.arm64-v8a client --log-level DEBUG get-mess-storage
error: unrecognized subcommand 'get-mess-storage'

  tip: some similar subcommands exist: 'set-mass-storage', 'get-mass-storage'

Usage: msd-tool.arm64-v8a client [OPTIONS] <COMMAND>

For more information, try '--help'.
2|kali:/data/adb/modules/com.chiller3.msd # ./msd-tool.arm64-v8a client --log-level DEBUG get-mass-storage
2024-11-23T05:34:40.966395Z ERROR msd_tool: Failed to receive protocol version acknowledgement

Caused by:
    Connection reset by peer (os error 104)

debianroot20200524, 2024-11-23:
[image]

chenxiaolong, 2024-11-23:
Please zip up `/data/local/tmp/msd/` and upload that here. I can take a look when I'm back from vacation in a couple days.

debianroot20200524, 2024-11-23:
kali:/ # cat /data/local/tmp/msd/msd-tool.log
--------- beginning of main
11-23 12:26:59.253  2254 20423 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:26:59.254  2254 20423 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:07.813  2254 20679 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:07.813  2254 20679 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:20.551  2254 21145 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:20.551  2254 21145 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:20.554  2254 21146 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:20.554  2254 21146 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:21.559  2254 21150 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:21.559  2254 21150 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:23.997  2254 21152 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:24.004  2254 21152 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:31.566  2254 21155 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:31.566  2254 21155 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:32.809  2254 21156 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:32.813  2254 21157 I msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:27:32.813  2254 21157 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:27:32.814  2254 21156 E msd-tool: peer{pid=20402 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:36:36.210  2254 22921 I msd-tool: peer{pid=22920 uid=0 gid=0}: msd_tool::daemon: Received connection
11-23 12:36:36.210  2254 22921 E msd-tool: peer{pid=22920 uid=0 gid=0}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:38:46.312  2254 23257 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:38:46.312  2254 23257 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:37.159  2254 23885 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:37.160  2254 23885 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:37.170  2254 23886 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:37.170  2254 23886 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:39.384  2254 23887 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:39.385  2254 23887 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:40.821  2254 23888 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:40.821  2254 23888 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:41.011  2254 23889 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:41.011  2254 23889 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:41.225  2254 23890 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:41.226  2254 23890 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:41.385  2254 23891 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:41.385  2254 23891 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 12:39:45.713  2254 23894 I msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 12:39:45.713  2254 23894 E msd-tool: peer{pid=23236 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
[... cut ...]
11-23 13:40:04.990  2254  7355 E msd-tool: peer{pid=7331 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
11-23 13:40:16.901  2254  7420 I msd-tool: peer{pid=7331 uid=10519 gid=10519}: msd_tool::daemon: Received connection
11-23 13:40:16.905  2254  7420 E msd-tool: peer{pid=7331 uid=10519 gid=10519}: msd_tool::daemon: Thread failed: Denying connection because SELinux is not enforcing
The problem has been solved.

MSD needs SELinux enforcing,but Kali Nethunter always set SELinux to permissive on boot.

closed 2024-11-23 by debianroot20200524