chenxiaolong/MSD #36: Failed to read directory: "/config/usb_gadget/g1/configs/b.1"  Caused by:     Permission denied (os error 13)

Anarchtism, 2025-01-01:
Similar issue to #19.
Running
`su -c "chown -R system:system /config/usb_gadget/g1"`
as [suggested](https://github.com/chenxiaolong/MSD/issues/19#issuecomment-2453303161) fixes the issue until a reboot, where it needs to be run again.

.zip containing everything in `/data/local/tmp/msd/` as [requested](https://github.com/chenxiaolong/MSD/issues/19#issuecomment-2567097304):
[msd.zip](https://github.com/user-attachments/files/18286259/msd.zip)

chenxiaolong, 2025-01-01:
Thanks for the logs!

This is the issue:
```
/config/usb_gadget:
total 0
drwxrwx---    6 root     system   u:object_r:configfs:s0                   0 Jan  1 10:13 g1
...
```
The only USB-related path that's writable by `system` on your device is `/config/usb_gadget/g1`. Everything else is owned by `root:root`.

However, that specific path happens to be what MSD checks for. Since it's writable, MSD never attempted to chown.

I'll work on improving the logic for this.

Anarchtism, 2025-01-01:
[image]
Interesting. Why, if I do `su -c 'ls -lZ /config/usb_gadget/g1'`, everything is root root, but I can only see that root system thing if I do `su -c 'ls -lZ /config/usb_gadget/'`?

Anarchtism, 2025-01-01:
It's like the main directory is writable by `system` and the sub-directories are not, I guess?

chenxiaolong, 2025-01-02:
> Why, if I do `su -c 'ls -lZ /config/usb_gadget/g1'`, everything is root root, but I can only see that root system thing if I do `su -c 'ls -lZ /config/usb_gadget/'`?

This is quirk of how `ls` works. It doesn't show the `g1` directory itself, only the contents. If you add `-d` (`ls -lZd /config/usb_gadget/g1`), then it does the opposite: show the directory, but not its contents.

> It's like the main `g1` directory is writable by `system` and the sub-directories are not, I guess?

Yep, exactly.

---

I've updated MSD to just always `chown` the directory instead of trying to guess when to do it. I don't think this change should break anything for other devices.

Please give #37 a try. Test build: [MSD-1.9.r2.g3a24f3b-release.zip](https://github.com/user-attachments/files/18287460/MSD-1.9.r2.g3a24f3b-release.zip)

Anarchtism, 2025-01-02:
Yep, that test build works!

closed 2025-01-02 by Anarchtism

chenxiaolong, 2025-01-02:
Perfect, thanks for testing! I'll release 1.10 with the fix later today.

PR #37 (2025-01-03): msd-tool: Unconditionally chown usb_gadget directory
This was previously done conditionally out of an abundance of caution in case some other process depends on the permissions not changing. There have been a number of cases where MSD's heuristic of when to chown isn't sufficient on real world devices, so let's just always chown.

In practice, there are only ever two situations:

* the USB gadget is managed by Android's USB gadget HAL, which runs as the `system` user
* the USB gadget is managed by init scripts, which run as `root`

Making everything owned by `system` should not break either sc [... cut]
commit 3a24f3b5 (2025-01-03): msd-tool: Unconditionally chown usb_gadget directory

This was previously done conditionally out of an abundance of caution
in case some other process depends on the permissions not changing.
There have been a number of cases where MSD's heuristic of when to chown
isn't sufficient on real world devices, so let's just always chown.

In practice, there are only ever two situations:

* the USB gadget is managed by Android's USB gadget HAL, which runs as
  the system user
* the USB gadget is managed by init scripts, which run as root

Making everything owned by system should not break either scena [... cut]