chenxiaolong/MSD #39: version for android 11?

Hellscaped, 2025-01-31:
Yoyoyo is there a way to get this on 11?

chenxiaolong, 2025-01-31:
Looks like there's no good reason to keep the minimum at Android 13. I can reduce it to Android 10 without needing any code changes.

Please give #40 a try. (Test build: [MSD-1.10.r2.g84c188b-release.zip](https://github.com/user-attachments/files/18613490/MSD-1.10.r2.g84c188b-release.zip))

Hellscaped, 2025-02-01:
> Looks like there's no good reason to keep the minimum at Android 13. ~I can reduce it to Android 10 without needing any code changes.~
> 
> Please give [#40](https://github.com/chenxiaolong/MSD/pull/40) a try. ~(Test build: [MSD-1.10.r2.g84c188b-release.zip](https://github.com/user-attachments/files/18613490/MSD-1.10.r2.g84c188b-release.zip))~
> 
> EDIT: Actually, minimal changes were needed. New test build: [MSD-1.10.r2.g6c01ca6-release.zip](https://github.com/user-attachments/files/18613935/MSD-1.10.r2.g6c01ca6-release.zip)

Thanks, I got a native crash error. Something something SELinux setting error.

`--------- beginning of crash
01-31 21:05:02.521  4265  4265 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 4265 (main), pid 4265 (main)
01-31 21:05:02.594  4524  4524 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-31 21:05:02.594  4524  4524 F DEBUG   : Native Crash TIME: 109465
01-31 21:05:02.594  4524  4524 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-31 21:05:02.594  4524  4524 F DEBUG   : Build fingerprint: 'BLU/G71_Plus/G0570WW:11/RP1A.201005.001/220:user/release-keys'
01-31 21:05:02.594  4524  4524 F DEBUG   : Revision: '0'
01-31 21:05:02.594  4524  4524 F DEBUG   : ABI: 'arm64'
01-31 21:05:02.604  4524  4524 F DEBUG   : Timestamp: 2025-01-31 21:05:02-0600
01-31 21:05:02.604  4524  4524 F DEBUG   : pid: 4265, tid: 4265, name: main  >>> zygote64 <<<
01-31 21:05:02.604  4524  4524 F DEBUG   : uid: 10251
01-31 21:05:02.604  4524  4524 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
01-31 21:05:02.604  4524  4524 F DEBUG   : Abort message: 'JNI FatalError called: (com.chiller3.msd) frameworks/base/core/jni/com_android_internal_os_Zygote.cpp:1776: selinux_android_setcontext(10251, 0, "default:privapp:targetSdkVersion=30:complete", "com.chiller3.msd") failed'
01-31 21:05:02.604  4524  4524 F DEBUG   :     x0  0000000000000000  x1  00000000000010a9  x2  0000000000000006  x3  0000007ff636ede0
01-31 21:05:02.604  4524  4524 F DEBUG   :     x4  000000752e043000  x5  000000752e043000  x6  000000752e043000  x7  000000000000aec8
01-31 21:05:02.604  4524  4524 F DEBUG   :     x8  00000000000000f0  x9  0000007529ea4840  x10 ffffff80fffffbdf  x11 0000000000000001
01-31 21:05:02.604  4524  4524 F DEBUG   :     x12 00000000ef4d0344  x13 478f00000012810c  x14 001efe91b3740d80  x15 00001137004e900a
01-31 21:05:02.604  4524  4524 F DEBUG   :     x16 0000007529f3cc80  x17 0000007529f1e420  x18 000000752da52000  x19 00000000000010a9
01-31 21:05:02.604  4524  4524 F DEBUG   :     x20 00000000000010a9  x21 00000000ffffffff  x22 0000007296ff337f  x23 0000000000000016
01-31 21:05:02.604  4524  4524 F DEBUG   :     x24 0000007296fd30b7  x25 0000000000000001  x26 0000007296fea0e1  x27 00000072975e2000
01-31 21:05:02.604  4524  4524 F DEBUG   :     x28 b4000072b764f690  x29 0000007ff636ee60
01-31 21:05:02.605  4524  4524 F DEBUG   :     lr  0000007529ed2148  sp  0000007ff636edc0  pc  0000007529ed2178  pst 0000000000000000
01-31 21:05:02.613  1425  1793 D performance: handleVmpressure, swap:31 pressure:1 thrashing:-1
01-31 21:05:02.751  4524  4524 F DEBUG   : backtrace:
01-31 21:05:02.751  4524  4524 F DEBUG   :     NOTE: Function names and BuildId information is missing for some frames due
01-31 21:05:02.751  4524  4524 F DEBUG   :     NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
01-31 21:05:02.751  4524  4524 F DEBUG   :     NOTE: found under the lib/ directory are readable.
01-31 21:05:02.751  4524  4524 F DEBUG   :       #00 pc 000000000004e178  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: ddedea5a1d1071f97c5321841b6be985)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #01 pc 0000000000552798  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort(char const*)+2260) (BuildId: 79991d5a5ba026c69376ba7be4e58e38)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #02 pc 0000000000013990  /system/lib64/libbase.so (android::base::SetAborter(std::__1::function<void (char const*)>&&)::$_3::__invoke(char const*)+76) (BuildId: 40363036c1f5a305d00fe1d058a86ccb)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #03 pc 0000000000012fb4  /system/lib64/libbase.so (android::base::LogMessage::~LogMessage()+320) (BuildId: 40363036c1f5a305d00fe1d058a86ccb)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #04 pc 00000000003fb840  /apex/com.android.art/lib64/libart.so (art::JNI<true>::FatalError(_JNIEnv*, char const*)+196) (BuildId: 79991d5a5ba026c69376ba7be4e58e38)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #05 pc 0000000000168390  /system/lib64/libandroid_runtime.so ((anonymous namespace)::ZygoteFailure(_JNIEnv*, char const*, _jstring*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)+160) (BuildId: 686462ce74a6252ebd5e68b65acadd43)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #06 pc 000000000016ae5c  /system/lib64/libandroid_runtime.so ((anonymous namespace)::SpecializeCommon(_JNIEnv*, unsigned int, unsigned int, _jintArray*, int, _jobjectArray*, long, long, int, _jstring*, _jstring*, bool, bool, _jstring*, _jstring*, bool, _jobjectArray*, _jobjectArray*, bool, bool)+8320) (BuildId: 686462ce74a6252ebd5e68b65acadd43)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #07 pc 000000000016675c  /system/lib64/libandroid_runtime.so (android::com_android_internal_os_Zygote_nativeForkAndSpecialize(_JNIEnv*, _jclass*, int, int,  [... cut ...]
01-31 21:05:02.751  4524  4524 F DEBUG   :       #20 pc 00000000000a0880  /system/lib64/libandroid_runtime.so (android::AndroidRuntime::start(char const*, android::Vector<android::String8> const&, bool)+836) (BuildId: 686462ce74a6252ebd5e68b65acadd43)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #21 pc 0000000000003574  /system/bin/app_process64 (main+1324) (BuildId: 6198d66333afe419e25cf31b1a39e03c)
01-31 21:05:02.751  4524  4524 F DEBUG   :       #22 pc 000000000004973c  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+108) (BuildId: ddedea5a1d1071f97c5321841b6be985)`

chenxiaolong, 2025-02-01:
Hmm, that's odd. Can you send me a link to the ROM you're running?

I wonder if older Android versions aren't respecting `/system/etc/selinux/plat_seapp_contexts`.

chenxiaolong, 2025-02-01:
Could you also zip up the contents of `/data/local/tmp/msd` and upload that?

Hellscaped, 2025-02-01:
> Hmm, that's odd. Can you send me a link to the ROM you're running?
> 
> I wonder if older Android versions aren't respecting `/system/etc/selinux/plat_seapp_contexts`.

I'm actually running pretty much stock AOSP. My phone is from BLU and they don't do anything special.

Hellscaped, 2025-02-01:
[msd.zip](https://github.com/user-attachments/files/18630214/msd.zip)

directory as requested

Hellscaped, 2025-02-01:
> Hmm, that's odd. Can you send me a link to the ROM you're running?
> 
> I wonder if older Android versions aren't respecting `/system/etc/selinux/plat_seapp_contexts`.

the file here HAS an entry for MSD. No clue whats happening with it though.

Hellscaped, 2025-02-01:
https://source.android.com/docs/core/permissions/perms-allowlist

Could it be something to do with this?

Hellscaped, 2025-02-01:
my root method is through APatch- KSU doesn't support my device and Magisk bootloops on my chipset.

Hellscaped, 2025-02-01:
> my root method is through APatch- KSU doesn't support my device and Magisk bootloops on my chipset.

I don't think APatch would be the issue here as the module system is based on KernelSU so KSU modules are compatible with it.

chenxiaolong, 2025-02-02:
> [msd.zip](https://github.com/user-attachments/files/18630214/msd.zip)
> 
> directory as requested

Thanks for the logs. Looks like this is the issue (from `post-fs-data.log`):
```
2025-02-01T03:03:23.036439Z ERROR msd_tool: Type not found: usb_control_prop
```

Older versions of Android are missing `usb_control_prop` and all userfaultfd SELinux types. I've made some changes to account for this.

Can you try this new test build? [MSD-1.10.r2.gb1de579-release.zip](https://github.com/user-attachments/files/18630795/MSD-1.10.r2.gb1de579-release.zip) If it still has issues, please attach a new zip of `/data/local/tmp/msd`.

Hellscaped, 2025-02-03:
> Thanks for the logs. Looks like this is the issue (from `post-fs-data.log`):
>
> ```
> 2025-02-01T03:03:23.036439Z ERROR msd_tool: Type not found: usb_control_prop
> ```
>
> Older versions of Android are [...]

New issue, app starts but I can't enable the devices.

[msd.zip](https://github.com/user-attachments/files/18637445/msd.zip)

funny thing is, my manu had the gadget and everything here so I didn't need to compile the kernel myself. Theres apparently some official version with some iso in the vendor dir but no UI for it.

chenxiaolong, 2025-02-04:
Looks like SELinux is all good now. For the new issue, could you grab the logs after trying to enable the devices? (`msd-tool.log` is empty in the latest upload--it should have some errors about what failed.)

Hellscaped, 2025-02-04:
```
--------- beginning of main
--------- beginning of system
--------- beginning of crash
02-02 23:16:51.816  1184  9162 I msd-tool: peer{pid=9132 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-02 23:16:56.306  1184  9189 I msd-tool: peer{pid=9132 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-02 23:16:56.317  1184  9189 E msd-tool: peer{pid=9132 uid=10251 gid=10251}: msd_tool::daemon: Thread failed: Failed to receive request
02-02 23:16:56.345  1184  9190 I msd-tool: peer{pid=9132 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-03 10:56:18.035  1184 30030 I msd-tool: peer{pid=29998 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 12:02:51.878  1184 14261 I msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 12:03:08.954  1184 14450 I msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 12:03:08.964  1184 14450 E msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Thread failed: Failed to receive request
02-04 12:03:08.978  1184 14451 I msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 12:03:15.361  1184 14484 I msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 12:03:15.370  1184 14484 E msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Thread failed: Failed to receive request
02-04 12:03:15.385  1184 14485 I msd-tool: peer{pid=14233 uid=10251 gid=10251}: msd_tool::daemon: Received connection

```

chenxiaolong, 2025-02-04:
Ah shoot, the logging is broken and doesn't show the root cause of `Thread failed: Failed to receive request`. Can you grab the logs one more time with this version? [MSD-1.10.r2.g4fd083a-release.zip](https://github.com/user-attachments/files/18661751/MSD-1.10.r2.g4fd083a-release.zip)

Hellscaped, 2025-02-04:
> Ah shoot, the logging is broken and doesn't show the root cause of `Thread failed: Failed to receive request`. Can you grab the logs one more time with this version? [MSD-1.10.r2.g4fd083a-release.zip](https://github.com/user-attachments/files/18661751/MSD-1.10.r2.g4fd083a-release.zip)

Rebooting right now, will send logs

Hellscaped, 2025-02-04:
--------- beginning of main
--------- beginning of system
--------- beginning of crash
02-04 16:25:47.330  1145  5916 I msd-tool: peer{pid=5819 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 16:25:49.885  1145  6075 I msd-tool: peer{pid=5819 uid=10251 gid=10251}: msd_tool::daemon: Received connection
02-04 16:25:49.912  1145  6075 E msd-tool: peer{pid=5819 uid=10251 gid=10251}: msd_tool::daemon: Thread failed: Failed to receive request
02-04 16:25:49.912  1145  6075 E msd-tool: 
02-04 16:25:49.912  1145  6075 E msd-tool: Caused by:
02-04 16:25:49.912  1145  6075 E msd-tool:     Ancillary data has no message
02-04 16:25:49.937  1145  6077 I msd-tool: peer{pid=5819 uid=10251 gid=10251}: msd_tool::daemon: Received connection

chenxiaolong, 2025-02-05:
Wow, that is not what I expected to see. I have absolutely no idea what would cause that. When it fails, does the UI show the exact same error message as in the logs?

I'll probably have to try and get MSD running in an emulator to debug this.

Hellscaped, 2025-02-05:
> Wow, that is not what I expected to see. I have absolutely no idea what would cause that. When it fails, does the UI show the exact same error message as in the logs?
> 
> I'll probably have to try and get MSD running in an emulator to debug this.

It shows this.

[image]

chenxiaolong, 2025-02-07:
I'm able to successfully reproduce the issue in an emulator running Android 11. I'll see if I can figure out what is going on here.

Hellscaped, 2025-02-07:
> I'm able to successfully reproduce the issue in an emulator running Android 11. I'll see if I can figure out what is going on here.

o7 man, just got an email (forgot to turn off DND)

chenxiaolong, 2025-02-07:
Please give this new test build a try: [MSD-1.10.r2.g4be5e09-release.zip](https://github.com/user-attachments/files/18701990/MSD-1.10.r2.g4be5e09-release.zip)

Hellscaped, 2025-02-07:
Works 🤟

Tested with a desktop and I get a prompt to format the drive.

What was the issue?

chenxiaolong, 2025-02-08:
Awesome, thanks for testing!

> What was the issue?

When you select a file from the system file picker, the file descriptor can be opened from multiple sources:

* the downloads provider (`com.android.providers.downloads`)
  * Used when selecting anything from the `Downloads` sidebar option (but not when manually navigating to the downloads directory)
* the media provider (`com.android.providers.media.module`)
  * Used when selecting anything from the documents, images, and videos sidebar options
* the external storage provider (`com.android.externalstorage`)
  * Used when selecting anything from the internal storage sidebar option

Selecting files from the downloads and media providers worked fine, but selecting a file from the external storage provider was broken. On Android 11, file descriptors opened by the external storage provider have the `platform_app` SELinux context (vs. `mediaprovider_app` on newer Android versions). The SELinux policy did not have a rule to allow the MSD app to pass a `platform_app` file descriptor to the MSD daemon.

This is what resulted in `Ancillary data has no message` message in the logs (ancillary data is how file descriptors get sent between processes via Unix sockets). When this happens, the request from the app -> daemon gets interrupted and there's no way to reply with a sensible response. The daemon terminates the connection, leading to the `Connection reset by peer` error you saw in the app.

Adding the one missing SELinux rule fixed the problem.

chenxiaolong, 2025-02-08:
I've just merged #40. I'll release MSD 1.11 later today after a couple more changes.

closed 2025-02-08 by chenxiaolong

lannibu, 2025-07-24:
Experiencing quite similar issues here but on Android 13. CLI tool works absolutely fine without issues

[image]

[image]

[image]

If any additional logs are required I would provide

commit 6c01ca68 (2025-01-31): Reduce minimum API level to 29

This is the lowest we can go with minimal code changes.

Closes: #39
commit b1de5791 (2025-02-02): Reduce minimum API level to 29

This is the lowest we can go with minimal code changes.

Closes: #39
commit 4fd083a1 (2025-02-04): Reduce minimum API level to 29

This is the lowest we can go with minimal code changes.

Closes: #39
commit 90881554 (2025-02-07): Reduce minimum API level to 30

The biggest changes are for the SELinux policy:

* sys.usb.controller is labelled as exported2_system_prop
* userfaultfd types do not exist
* com.android.externalstorage.documents file descriptors are labeled as
  platform_app

Closes: #39