chenxiaolong/MSD #42: More free storage

alr86, 2025-02-22:
I want to install Batocera on my phone's MSD. Can I increase the storage to save more ROMs on it?

chenxiaolong, 2025-02-22:
Sorry, I don't understand what you're asking for.

Do you mean you already have an existing image and you want to resize it? If so, this is not supported in MSD. I can maybe add a feature for changing the size of the image file, but it wouldn't be too useful because you would still need to manually resize the partitions and filesystems inside of it.

alr86, 2025-02-23:
That's what I meant 👍

chenxiaolong, 2025-02-23:
Awesome, thanks for clarifying! I've added support for resizing writable disk images in #43. It'll be included in MSD 1.12, which I plan to release later today.

commit 980c4ac4 (2025-02-23): Add support for resizing writable disk images

This only changes the size of the image file. The user is responsible
for manually repartitioning the contents if needed.

Fixes: #42
PR #43 (2025-02-23): Add support for resizing writable disk images
This only changes the size of the image file. The user is responsible for manually repartitioning the contents if needed.

Fixes: #42
closed 2025-02-23 by chenxiaolong