chenxiaolong/MSD #45: MTP and USB image access stops working until you restart the phone

DvaMishkiLapa, 2025-03-06:
Good afternoon, thank you very much for your project!
I don't know how to describe the problem, that's why the title is so general.

I have OnePlus 7 Android 15 Magisk 28.1 (Zygisk enabled), MSD 1.12.
```sh
# uname -a
Linux localhost 4.14.336 #1 SMP PREEMPT Tue Feb 25 23:51:53 IST 2025 aarch64 Toybox

# ls /config/usb_gadget/g1
UDC  bDeviceClass  bDeviceProtocol  bDeviceSubClass  bMaxPacketSize0  bcdDevice  bcdUSB  configs  functions  idProduct  idVendor  os_desc  strings

# zcat /proc/config.gz | grep CONFIG_USB_CONFIGFS_MASS_STORAGE
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
```

I created a test `img` file, formatted it as exFAT. It is in the internal drive. Attempting to connect breaks MTP until the device reboots.

My steps for replay:
1. Reboot phone;
2. Connect phone to PC by USB-C Cable. MTP is working;
3. Select my `.img` file by Writable disk and press Apply Settings. Now MTP doesn't work. There's no PC access to my image either.;

- Trying to reconnect the cable before the phone reboots does nothing;
- Enabling USB debugging had no effect whatsoever.

Since USB debugging also breaks in the process, I am attaching all `log` files from the msd folder.
[ps.log](https://github.com/user-attachments/files/19106708/ps.log)
[service.log](https://github.com/user-attachments/files/19106709/service.log)
[configfs.log](https://github.com/user-attachments/files/19106706/configfs.log)
[dmesg.log](https://github.com/user-attachments/files/19106712/dmesg.log)
[mountinfo.log](https://github.com/user-attachments/files/19106707/mountinfo.log)
[msd-tool.log](https://github.com/user-attachments/files/19106711/msd-tool.log)
[post-fs-data.log](https://github.com/user-attachments/files/19106710/post-fs-data.log)

Before the USB connection failed, in the debug console I saw this:
```sh
adb logcat -v color -s msd-tool

--------- beginning of main
03-06 11:02:07.733  1999 11357 I msd-tool: peer{pid=11336 uid=10277 gid=10277}: msd_tool::daemon: Received connection
03-06 11:13:59.749  1999 11910 I msd-tool: peer{pid=11336 uid=10277 gid=10277}: msd_tool::daemon: Received connection
03-06 11:13:59.829  1999 11910 W msd-tool: peer{pid=11336 uid=10277 gid=10277}: msd_tool::daemon: No gadget HAL process found: android.hardware.usb.gadget-service*
03-06 11:13:59.829  1999 11910 D msd-tool: peer{pid=11336 uid=10277 gid=10277}: msd_tool::daemon: Disassociating gadget config from controller
```

DvaMishkiLapa, 2025-03-06:
Interestingly, if I reboot the phone with USB-C plugged in, enable USB mode on File Transfer, and then in MSD select my image and click Apply Settings in MSD, everything works fine and I can see the disk image on the PC.
But after reconnecting the USB-C MTP and accessing the image from the PC, it breaks again.

[reboot_ps.log](https://github.com/user-attachments/files/19107177/reboot_ps.log)
[reboot_service.log](https://github.com/user-attachments/files/19107176/reboot_service.log)
[reboot_configfs.log](https://github.com/user-attachments/files/19107174/reboot_configfs.log)
[reboot_dmesg.log](https://github.com/user-attachments/files/19107173/reboot_dmesg.log)
[reboot_msd-tool.log](https://github.com/user-attachments/files/19107178/reboot_msd-tool.log)
[reboot_mountinfo.log](https://github.com/user-attachments/files/19107175/reboot_mountinfo.log)
[reboot_post-fs-data.log](https://github.com/user-attachments/files/19107196/reboot_post-fs-data.log)

chenxiaolong, 2025-03-06:
Thanks for the logs! When you reconnect the USB cable and encounter the problem, what does MSD report in the `Active USB gadget functions` section? (You can tap it to refresh, just in case) With MTP + ADB + mass storage enabled, I would expect it to show `ffs.mtp`, `ffs.adb`, and `mass_storage.msd` in there.

If all 3 are showing up after the issue happens, then the USB controller is still configured properly and you're most likely hitting a kernel bug. Unfortunately, I don't really have any suggestions for what to do if that ends up being the case.

One thing that might be worth trying is running `su -c 'dmesg -w'` in Termux and seeing if it logs any new messages when reconnecting the USB cable. If the phone is actually detecting a connection, it should definitely print something out.

DvaMishkiLapa, 2025-03-07:
So, here's what I did:
1. Rebooted the phone, connected to PC, enabled File Transfer, opened MSD and connected the disk. Everything works and shows up in PC.
2. I reconnected the cable to the PC. The disk and MTP device is gone. I re-enabled File Transfer. In MSD I see the following:
   ```
   f1: mtp.gs0
   f2: ffs.adb
   msd: mass_storage.msd
   ```
   But the disk and MTP device does not appear in PC. The phone shows me that there is a USB connection, but on the PC side I don't 
   see any devices. The Active USB gadget list is the same as in step 2.
4. In MSD I clicked Apply settings and the phone stopped seeing that there is a cable connected to it. The Active USB gadget list is the same as in step 2.
5. After 3 minutes the phone saw the cable connection. I did not disconnect it the whole time. The Active USB gadget list is the same as in step 2.

Interestingly, in step 2 and step 4 ADB works. But MTP and MSD are not working.

[dmesg.log](https://github.com/user-attachments/files/19124400/dmesg.log)
- `[   70.123484] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?` - Too bad, but no. The cable is serviceable and I have tried others;
- `[   93.701866] android_work: sent uevent USB_STATE=DISCONNECTED` - I think I unplugged the cable somewhere around here.

If there's anything else I can do, tell me.
If you see any kernel problem, I can try to contact the maintainer of that kernel and try to find out the details.

chenxiaolong, 2025-03-07:
Thanks for attaching the full dmesg output. It definitely looks like there's a kernel issue. There are 9 crashes like this in the mass storage kernel driver:

```
[  339.587458] configfs-gadget gadget: common->fsg is NULL in get_next_command at 2436
[  339.587487] ------------[ cut here ]------------
[  339.587507] WARNING: CPU: 5 PID: 9192 at fsg_main_thread+0x135c/0x4930
[  339.587513] CPU: 5 PID: 9192 Comm: file-storage Tainted: G S      W       4.14.336 #1
[  339.587515] Hardware name: Qualcomm Technologies, Inc. SM8150 V2 PM8150 MTP 18857 14 15 (DT)
[  339.587518] task: 000000006c96b201 task.stack: 000000006452741d
[  339.587522] pc : fsg_main_thread+0x135c/0x4930
[  339.587525] lr : fsg_main_thread+0x135c/0x4930
[  339.587527] sp : ffffffea9098bd40 pstate : 60c00145
[  339.587528] x29: ffffffea9098be10 x28: 0000000000000004 
[  339.587532] x27: ffffff8a19a0bc50 x26: ffffffea9098bd98 
[  339.587535] x25: ffffffea80923c30 x24: ffffffea44892f00 
[  339.587537] x23: ffffffe9dbf73400 x22: 0000000000000000 
[  339.587539] x21: ffffffea80923c38 x20: ffffffea050b7a78 
[  339.587541] x19: ffffffea050b7a00 x18: 0000000000000000 
[  339.587543] x17: 0000000000000080 x16: 0000000000000080 
[  339.587545] x15: ffffff8a1a30215c x14: 0000000000000018 
[  339.587547] x13: 00000000341555ad x12: 0000000000000000 
[  339.587549] x11: 0000000000000000 x10: ffffffffffffffff 
[  339.587551] x9 : 1a1c366573ef0900 x8 : 1a1c366573ef0900 
[  339.587553] x7 : 0720072007200720 x6 : ffffffeb76880f00 
[  339.587555] x5 : 0a0a0a0a0a0a0a0a x4 : 0000000000000000 
[  339.587557] x3 : 0000000000000000 x2 : 0000000000000001 
[  339.587559] x1 : 0000000000000000 x0 : 0000000000000047 
[  339.587564] \x0aPC: fsg_main_thread+0x131c/0x4930:
[  339.587565] 279c  91002328 88dffd08 37fffde8 91006260 910103e1 97cf7b48 f9400a76 b5ffead6
[  339.587574] 27bc  f9400268 91018100 900079c1 913c4c21 f0007742 91114042 52813083 97f055c2
[  339.587581] 27dc  d4210000 b9000b3f 17fffb52 b90043ff a904eff7 f9002ffa f90033fa 91006260
[  339.587588] 27fc  910103e1 52800022 97cf7a5b b50000e0 9432a0e9 91006260 910103e1 52800022
[  339.587596] \x0aLR: fsg_main_thread+0x131c/0x4930:
[  339.587597] 279c  91002328 88dffd08 37fffde8 91006260 910103e1 97cf7b48 f9400a76 b5ffead6
[  339.587603] 27bc  f9400268 91018100 900079c1 913c4c21 f0007742 91114042 52813083 97f055c2
[  339.587609] 27dc  d4210000 b9000b3f 17fffb52 b90043ff a904eff7 f9002ffa f90033fa 91006260
[  339.587616] 27fc  910103e1 52800022 97cf7a5b b50000e0 9432a0e9 91006260 910103e1 52800022
[  339.587627] \x0aSP: 0xffffffea9098bd00:
[  339.587628] bd00  18d627dc ffffff8a 60c00145 00000000 ffffffd0 ffffff80 19c9af13 ffffff8a
[  339.587634] bd20  ffffffff ffffffff 73ef0900 1a1c3665 9098be10 ffffffea 18d627dc ffffff8a
[  339.587641] bd40  00000000 00000000 dbf73400 ffffffe9 76936800 ffffffeb 1a0ef598 ffffff8a
[  339.587647] bd60  00000002 00000000 050b7bc0 ffffffea 050b7ab9 ffffffea 9098bd98 ffffffea
[  339.587654] 
[  339.587656] Call trace:
[  339.587659] fsg_main_thread+0x135c/0x4930
[  339.587662] kthread+0x19c/0x1b0
[  339.587665] ret_from_fork+0x10/0x18
[  339.587667] ---[ end trace 9334e44e8b629a83 ]---
```

DvaMishkiLapa, 2025-03-09:
Yes, it's a problem on the kernel side. Thank you very much for your help.

closed 2025-03-09 by DvaMishkiLapa