chenxiaolong/MSD #47: [HELP] Is there a way to access the virtual disk on the phone?

ztmaev, 2025-03-15:
**Disk info**
> writable disk:   /mnt/user/0/emulated/0/disk.img
> f1: ffs.adb
> msd: mass_storage.msd

Is there a way to mount the disk and access it directly on my phone's file explorer?

chenxiaolong, 2025-03-15:
There's no easy way to do this, but if your device's kernel is compiled with support for reading the filesystems on your virtual disk, then you can do it in Termux:

1. Install `partx`:

   ```bash
   pkg install blk-utils
   ```

2. Create loop devices for each partition in the image:

   ```bash
   su -c 'partx -av /sdcard/disk.img
   ```

3. Find the block device that was just created:

   ```bash
   su -c 'losetup -a`
   ```

   It will print out something like this:

   ```bash
   1|komodo:/ # losetup -a
   /dev/block/loop46p1: [161]:217162 (/storage/emulated/0/disk.img)
   /dev/block/loop46: [161]:217162 (/storage/emulated/0/disk.img)
   ```

   In this example, `loop46` is the entire disk and `loop46p1` is the first partition on the disk.

4. Mount the block device. For example:

   ```bash
   su -c 'mkdir /mnt/disk'
   su -c 'mount /dev/block/loop46p1 /mnt/disk'

Then you browse and modify the files in `/mnt/disk` with a root file manager.

To unmount the disk afterwards:

```bash
su -c 'umount /mnt/disk'
su -c 'losetup -d /sdcard/disk.img'
```

closed 2025-03-15 by chenxiaolong

e-t-l, 2026-09-06:
Can the img file be accessed through RSAF using a "local disk" type mount?

chenxiaolong, 2026-09-06:
> Can the img file be accessed through RSAF using a "local disk" type mount?

Sadly not. rclone only supports reading squashfs filesystems (which are read-only). It does not support any other Linux filesystems.